Xyce/Xyce #148: Xyce multiple plugins support + potential solution

KrzysztofHerman, 2025-06-25:
I am facing a strange behavior of Xyce while trying to run it with some ADMS compiled plugins. 
While referencing a comma separated plugin  location list I am able to run an example only when one or two plugins are listed. If I add a third one it causes error.  My netlist only needs `$PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_r3_cmc.so` but in the PDK we would like to use all available plugin list. 

The following configurations works:

```
Xyce -plugin  $PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_r3_cmc.so dc_res_temp.spice
Xyce -plugin  $PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_PSP103_VA.so,$PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_r3_cmc.so dc_res_temp.spice
Xyce -plugin  $PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_mosvar.so,$PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_r3_cmc.so dc_res_temp.spice

```
This one crashes
```
Xyce -plugin  $PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_mosvar.so,$PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_PSP103_VA.so,$PDK_ROOT/ihp-sg13g2/libs.tech/xyce/plugins/Xyce_Plugin_r3_cmc.so dc_res_temp.spice
``` 

I have successfully tested the following solution changing the lines 895-903 in `src/CircuitPKG/N_CIR_Xyce.C`
from this:
```
if (commandLine_.argExists(std::string("-plugin")))
  {
    const std::string plugin = commandLine_.getArgumentValue("-plugin");
    
    for (std::string::size_type i = 0, j = plugin.find_first_of(", "); i != std::string::npos; i = (j == std::string::npos ? j : j + 1), j = plugin.find_first_of(", ", i))
    {
      Device::registerPlugin(plugin.substr(i, j).c_str());
    }
  }
```
to this:

```
if (commandLine_.argExists(std::string("-plugin")))
{
  const std::string plugin = commandLine_.getArgumentValue("-plugin");

  std::string::size_type i = 0;
  std::string::size_type j = plugin.find_first_of(", ");
  while (j != std::string::npos)
  {
    if (j > i)
      Device::registerPlugin(plugin.substr(i, j - i).c_str());
    i = j + 1;
    j = plugin.find_first_of(", ", i);
  }
  // Handle the last item
  if (i < plugin.length())
    Device::registerPlugin(plugin.substr(i).c_str());
}
```

fnordpole, 2025-06-26:
A also fixed this bug in 2023 and sent them the code but since then nothing happened apparently
https://groups.google.com/g/xyce-users/c/_pGn9IuMLJM/m/-AapIgPDBAAJ

Karlsefni2012, 2025-06-26:
Well, that is embarrassing.  I am the person who dropped the ball on this issue.  I will make sure it gets fixed this time!

KrzysztofHerman, 2025-06-26:
@Karlsefni2012 @fnordpole thanks for your feedback. I am not s software developer so the provided code was generated, please review it carefully. The IHP-Open-PDK provides some Verilog-A models and we are heading into this direction so naturally we will have more plugins available. It is a pity that ADMS is not supported.

Karlsefni2012, 2025-06-28:
> FWIW, I have a nice clean fix for this that I will submit as a PR shortly.  Just testing it out and crafting a regression test for it.
> 

Thanks! I will apply your patches.

closed 2025-07-21 by jdschic

eskiyerli, 2026-04-18:
I still see this problem with Xyce version 7.10:
 Xyce -plugin "$PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_PSP103_VA.so, $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_mosvar.so, $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_r3_cmc.so" tb_inverter_tran.sp
Netlist error: Failed to load plugin
  $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_mosvar.so,
 /: cannot open shared object file: No such file or directory
Netlist error: Failed to load plugin
 $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_mosvar.so,
 $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_r3_cmc.so:
 cannot open shared object file: No such file or directory
Netlist error: Failed to load plugin
  $PROJECTS_ROOT/design_software/ihp_pdk/va_modules/libXyce_Plugin_r3_cmc.so:
 cannot open shared object file: No such file or directory
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 3 MSG_ERROR
 errors
*** Xyce Abort ***
Simulation aborted due to error.  There are 0 MSG_FATAL errors and 3 MSG_ERROR
 errors

tvrusso, 2026-04-20:
Oh, no.

The commit 40f2ae39a was introduced into Xyce *after* release 7.10, so it's no surprise that 7.10 still has the issue.

You'll have to build the code off of the master branch to get this fix.  Or wait until release 7.11, but I have no idea when the Xyce team intends to release that.