Xyce/Xyce #150: Xyce/ADMS problem with switched contribution

dwarning, 2025-06-29:
The r3_cmc Verilog-A model has following construct for modelling the end resistors:

`  

        if ((rc1_tnom/mMod)<=rthresh) begin    // V=R*I formulation

            V(b_re1) <+  I(b_re1)*rc1_tnom*tcrc;

        end else begin                         // I=G*V formulation

            I(b_re1) <+  V(b_re1)/(rc1_tnom*tcrc);

        end
`

This goes wrong in any case - means using the V or I formulation.
But using the contributions without the conditional switch works very well:

` 
V(b_re1) <+  I(b_re1) * rc1_tnom * tcrc;

`
The Xyce ADMS user guide didn't forbid either V nor I contributions. Only mixing both is not allowed.

Is this a known limitation of Xyce/ADMS? Is it an Xyce or ADMS issue?

tvrusso, 2025-06-29:
I should also add that there is nothing *preventing* Xyce/ADMS from being extended to support switch branches, and just prior to my retiring I was thinking about how to do it.  I just never got there, and got my sweet release from working on ADMS instead.

Xyce's capacitor model happens to have code that is somewhat similar to what would have to be emitted by Xyce/ADMS in support of switch branches --- Xyce's capacitor, when IC= is given, acts as if there is a voltage source across the capacitor that turns off after the DCOP.  This requires that different jacobians be loaded for the DCOP than at transient, and that the branch equation for the voltage source be reduced to something trivial during transient.

Switch branches would require the same sort of code to be emitted, but instead of being conditioned on transient vs. DCOP it'd have to be conditioned on something that can change any time during the run.

Doing this would be possible, but because ADMS is always difficult to work with in the best of cases, it would be a pretty big hack.

Hopefully, XyDeCo will not be written with this sort of limitation.

dwarning, 2025-06-29:
Thank you for the response.
Problem is the conditional switching.
Both contributions are accepted and bring correct results.
        V(b_re1) <+  I(b_re1)*rc1_tnom*tcrc;
        I(b_re1) <+  V(b_re1)/(rc1_tnom*tcrc);

tvrusso, 2025-06-29:
Exactly.  The problem is that when the voltage contribution is active (because of the conditional), Xyce must solve a problem with an extra equation and variable (the branch current and an equation to go with it), but when the current contribution is active it must make that extra equation be trivial and the extra variable irrelevant.

Xyce/ADMS has not been written to emit code to handle that.