Xyce/Xyce #158: BSIM-CMG parameter EOTACC has wrong bounds

itsayellow, 2025-07-29:
I'm simulating with Xyce using the ASAP7 models (https://github.com/The-OpenROAD-Project/asap7).  Many of the transistor models contain the parameter EOTACC set at 1.0e-010.

Xyce flags these incorrectly as out of bounds.
```
 Device model NMOS_RVT: ADMSbsimcmg: Parameter EOTACC value 1e-10 out of range
 [ 0.1*1.0e-9,  (+inf) [
```

The actual bounds in the BSIM-CMG manual (any version, AFAIK) specify that EOTACC has a minimum value of 0.1e-9, the same value as ASAP7 sets it at.

I believe this is just a simple case of the bounds checking code using open bounds (endpoints not included) when it should be using closed bounds (endpoints included).  

Other parameters in the BSIM-CMG manuals include a greater-than symbol for their minimums when they need to specify minimums without endpoints included.  EOTACC only specifies the minimum value, which I believe means that minimum value should also be valid.

tvrusso, 2025-07-29:
The EOTACC parameter is defined by the BSIM-CMG verilog-a source code as having an closed-bound range on the low side and open on the right.  These are correctly translated into C++ as ">=" for the closed endpoint.

BSIM-CMG 107: `parameter real EOTACC    =  EOT from [0.1n:inf);`

BSIM-CMG 108: `parameter real EOTACC    =  EOT from [0.1n:inf);`

BSIM-CMG 110:`` `MPRco( EOTACC         ,EOT            ,"m"           ,1.0e-10     ,inf         ,"Equivalent oxide thickness for accumulation region" )``
 (MPRco is the CMC standard macro for defining a real, closed-bound at lower end, open on upper, model parameter

BSIM-CMG-111:`` `MPRco(EOTACC, EOT, "m", 1.0e-10, inf, "Equivalent oxide thickness for accumulation region")``

These are translated into Xyce as (for example, from the BSIM-CMG 111):
```
//    Parameter EOTACC : [ 1.0e-10,  (+inf) [
  if ( (!((EOTACC >=1.0e-10))) )
  {
    UserWarning(*this) << "ADMSbsimcmg_111: Parameter EOTACC value " << EOTACC << " out of range [ 1.0e-10,  (+inf) [";
  }
```

The test is exactly "greater than or equal to 1e-10" and it should be accepting 1e-10 if that is the parameter value given.

The only thing I can guess is that there is some sort of rounding issue so that the ASAP7 model is not literally feeding "1e-10" but some expression that evaluates to slightly less than that and rounds to 1e-10, so strict bounds checking fails even with the ">=" in the test.

Since this is only a warning and not an error, it should be safe to ignore it.

tvrusso, 2025-07-29:
BTW, I also checked and found that if one replaces "1e-010" in the model card with "{0.1*1e-9}" the warning goes away.

itsayellow, 2025-07-29:
If changing the number in the model card to something numerically equivalent in decimal will fix the problem (and only change the value by ~ 1 part in 1e16 of the real value) this sounds like a good workaround to me.  At any rate it's down to machine precision.

Still it would be nice if there was a fix for this situation in general.  It's a pity that nobody is paying attention to ADMS, because for nm-size technologies it's unavoidable because they use BSIM-CMG.

tvrusso, 2025-07-29:
FWIW, what would be necessary to fix this is to modify the template "rangeCondition" in `xyceBasicTemplates_nosac.xml`, which is currently designed to turn a range expression in a parameter definition into a conditional term.  Currently, it generates `parameter >= bound` for each closed lower bound and `parameter > bound` for open lower bounds, and similarly for upper bounds.   This should probably get rewritten to emit `( parameter - (bound) >= machine precision)` and so forth.

This would not be a particularly difficult change.  One would have to:
     
   -    Modify `xyceImplementationFile_nosac.xml` to add an include of  `N_UTL_MachDepParams.h`, 
   -    Edit the rangeCondition template to emit a comparison of the difference between value and bound to `Util::MachineDependentParams::MachinePrecision()` instead of the direct comparison of value and bound.
   -   regenerate all the models (using `generate_ADMS.sh`, which I note was not actually updated to regenerate BSIM-CMG 111 when that model was added)
   -   Test the snot out of it to make sure it doesn't break any existing tests or spew even more meaningless warnings.

For example, where the rangeCondition template currently has:
```
      <admst:variable name="returnCondition" select="("/>
      <admst:if test="infexpr[hasspecialnumber!='YES']">
        <admst:variable name="returnCondition" select="$returnCondition%(../name) &gt;"/>
        <admst:if test="[infboundtype='range_bound_include']">
          <admst:variable name="returnCondition" select="$returnCondition="/>
        </admst:if>
        <admst:variable name="returnCondition" select="$returnCondition(%(processTerm(infexpr)/[name='returnedExpression']/value))"/>
        <admst:if test="supexpr[hasspecialnumber!='YES']">
          <admst:variable name="returnCondition" select="$returnCondition &amp;&amp; "/>
        </admst:if>
      </admst:if>
```
(which causes emission of a condition that reads `parameter > lowerbound`  for open or `parameter >= lowerbound` for closed ranges, and possibly emit `&&` after it if the upper bound must also be checked)

it would have to be something more like:
```
      <admst:variable name="returnCondition" select="("/>
      <admst:if test="infexpr[hasspecialnumber!='YES']">
        <admst:variable name="returnCondition" select="$returnCondition(%(../name) - (%(processTerm(infexpr)/[name='returnedExpression']/value) &gt;"/>
        <admst:if test="[infboundtype='range_bound_include']">
          <admst:variable name="returnCondition" select="$returnCondition="/>
        </admst:if>
        <admst:variable name="returnCondition" select="$returnCondition  Util::MachineDependentParams::MachinePrecision() )"/>
        <admst:if test="supexpr[hasspecialnumber!='YES']">
          <admst:variable name="returnCondition" select="$returnCondition &amp;&amp; "/>
        </admst:if>
      </admst:if>
```
so it would emit instead `( parameter - (lowerbound) > Util::MachineDependentParams::MachinePrecision() )` and so forth.  

This snippet adds a bunch of parens to the generated expression in an attempt to make sure things come out right in ways that might not have been important with the other format.  An  extra pair of parens around the lowerbound is to make sure that if the lower bound expression is actually doing arithmetic, the order of operations is done right.

A similar modification would have to be made for the similar section that follows for the upper bound, which differs from this block mainly by emitting `<` or `<=` instead and using "supexpr" instead of "infexpr" to get the bounding value.

This is a _completely_ untested code fragment I am pulling directly out of my lower torso and only applies to the lower bound.  I have not checked that my fragment does the parens correctly, but all it does is move the bound to the LHS of the comparison operator and put the machine precision on the RHS, so if the old one was right it should be OK.  I *have* added a paren around the bounds expression to make sure that any arithmetic that appears in that bounds gets properly evaluated, and if it isn't exactly right it  *should* be at least close to right.

It would then be required to regenerate all of the models in Xyce that originate from verilog-a to test that nothing breaks.   

Maybe someone on the team would be motivated to try this out for the fun of it.  

The rangeCondition template only processes a single range, and parameters can have more than one range (for example, they could include two separate segments of the real number line, but exclude a chunk in the middle, or only accept discrete values).  The multiple range thing is taken care of in other templates that call rangeCondition on each single range and assemble the collection of expressions into a single conditional statement.  Getting rangeCondition right should make everything else be OK with no modification.

ADMS is ancient technology and terribly limited, and has long been unsupported by its author.  The Qucs team took it over for a while, but it's not being maintained by them either as supposedly better tools exist now that they are using.  The Xyce team is also working on a replacement of its own, but it is still embryonic (as far as I know) and not ready to process complex models like the BSIM-CMG.  The team's staffing doesn't allow much effort to be expended on Xyce/ADMS these days.

tvrusso, 2025-07-29:
Alas, I'm talking nonsense here.

Comparing `value - bound >= machineprecision` isn't the right test at all.  It'll *always* fail if in fact you're already exactly at the lower bound so the difference is zero,  and it'll spew warnings like they're going out of style.

It was, however, simple to modify the ADMS back-end to generate it and let it tell me how much nonsense I was speaking.

A more careful condition is needed to take care of these edge cases.   And I've exhausted my free ADMS consultation budget.

tvrusso, 2025-07-29:
Actually, I changed my mind.  I'm not going to issue a pull request on this.

Making this change silences th warnings.  However, it is a blunt instrument because a lot of range checking is being done on integer parameters, and it's wasteful to be doing this sort of checking on *those*, but the ADMS code isn't refined enough to distinguish between integer and real parameters for the purpose of range checking.  

But as long as I check the correct sign of machine precision it does work to silence your warnings, however, so I'm attaching a patch that can be experimented with.  

Once Xyce/ADMS has been patched with this, one has to regenerate all the models of interest and copy the generated code back to the DeviceModelPKG/ADMS directory.  For example, one can do a "make all-source" in the Xyce/utils/ADMS/examples/`bsimcmg_107.0.0/code` directory, copy the generated `N_DEV_ADMSbsimcmg.C` file into the DeviceModelPKG/ADMS directory, and rebuild Xyce.   The bounds checking should no longer puke on input parameters that exceed the bounds by less than machine precision.

[patch_adms_range_check.txt](https://github.com/user-attachments/files/21496456/patch_adms_range_check.txt)

If the Xyce team cares to improve on that and adopt it, great.  Or not.