Xyce/Xyce #161: M_PI definition missing

dwarning, 2025-08-12:
I was able to build Xyce.exe under msys2 ucrt64 with Trilinos 16.1 and following cmake cmd:

` cmake -G "Unix Makefiles" -D CMAKE_INSTALL_PREFIX=/usr/local -D Trilinos_ROOT=$HOME/XyceLibs/serial -D FLEX_INCLUDE_DIR=/usr/include ..`

The onlyest quirk in source was a missing M_PI definition in N_UTL_BasicFFT_Interface.C which I solved by inserting the _USE_MATH_DEFINES macro before cmath. See diff below.

```
diff --git a/src/UtilityPKG/N_UTL_BasicFFT_Interface.C b/src/UtilityPKG/N_UTL_BasicFFT_Interface.C
index 247d24a0..207ae17a 100644
--- a/src/UtilityPKG/N_UTL_BasicFFT_Interface.C
+++ b/src/UtilityPKG/N_UTL_BasicFFT_Interface.C
@@ -36,7 +36,7 @@
 //
 //-------------------------------------------------------------------------
 // ---------- Standard Includes ----------
-
+#define _USE_MATH_DEFINES
 #include <Xyce_config.h>

 #include <N_UTL_BasicFFT_Interface.hpp>
```

tvrusso, 2025-08-12:
An alternate approach would be for `N_UTL_BasicFFT_Interface.C` and `N_UTL_AppleFFT_Interface.C` to include `N_UTL_Math.h` instead of cmath.

Those two files are the only two in Xyce that include `<cmath>` directly, all others get it from `N_UTL_Math.h`.   That file defines `_USE_MATH_DEFINES` *and* provides a fallback definition of `M_PI` in case cmath doesn't define it even with that macro declared.

Karlsefni2012, 2025-08-14:
I've made @tvrusso 's  change in our internal version.  It will show up here on github next time we do a sync.

tvrusso, 2025-08-27:
Fixed in commit c268934fc.

This issue could be closed now.

dwarning, 2025-08-28:
Checked it - it is OK now.
Thanks!

closed 2025-08-28 by dwarning