Xyce/Xyce #162: 7.10 release still marked as pre-release

tvrusso, 2025-08-12:
The 7.10 tag was pushed back in early July, 7.10 itself was published on the Xyce web site in early August, but the release on github is still marked pre-release.

I point this out only because I know that there is one system out there which keeps track of new Xyce releases and updates its Xyce port fairly soon thereafter, but they won't update if you don't publish the release officially on github.

jdschic, 2025-08-14:
Due to a delay in our internal release schedule for Xyce 7.10, we have not yet finalized all release activities. Despite the delay, we decided to make most of the resources for the new release available. The remaining items should wrap up in the next few days. Once the official announcement for Xyce 7.10 goes out, the pre-release status will be updated.

tvrusso, 2025-09-27:
Release 7.10 is no longer marked as pre-release, closing.

closed 2025-09-27 by tvrusso