Xyce/Xyce #164: ADMS compiled models stopped work in Xyce 7.10

KrzysztofHerman, 2025-08-19:
After installing Xyce 7.10 I can no longer run simulation with plugins.
`Xyce  -plugin ../../xyce/plugins/Xyce_Plugin_PSP103_VA.so dc_lv_nmos.spice`

The error is the following:
```
Netlist error: Failed to load plugin
 ../../xyce/plugins/Xyce_Plugin_PSP103_VA.so: undefined symbol:
 _ZTIN4Xyce6Device14DeviceInstanceE
```

My installation config is the following:
Trilinos
```
#!/bin/bash

FLAGS="-Wno-dev -O3 -fPIC"
cmake \
-DCMAKE_C_COMPILER=gcc \
-DCMAKE_CXX_COMPILER=g++ \
-DCMAKE_Fortran_COMPILER=gfortran \
-DCMAKE_CXX_FLAGS="$FLAGS" \
-DCMAKE_C_FLAGS="$FLAGS" \
-DCMAKE_Fortran_FLAGS="$FLAGS" \
-DCMAKE_INSTALL_PREFIX="/usr/local" \
-DCMAKE_MAKE_PROGRAM="make" \
-DTrilinos_ENABLE_NOX=ON \
-DNOX_ENABLE_LOCA=ON \
-DTPL_ENABLE_MPI=OFF \
-DTrilinos_ENABLE_EpetraExt=OFF \
-DEpetraExt_BUILD_BTF=ON \
-DEpetraExt_BUILD_EXPERIMENTAL=ON \
-DEpetraExt_BUILD_GRAPH_REORDERINGS=ON \
-DTrilinos_ENABLE_TrilinosCouplings=ON \
-DTrilinos_ENABLE_Ifpack=ON \
-DTrilinos_ENABLE_AztecOO=ON \
-DTrilinos_ENABLE_Belos=ON \
-DTrilinos_ENABLE_Teuchos=ON \
-DTrilinos_ENABLE_COMPLEX_DOUBLE=ON \
-DTrilinos_ENABLE_Amesos=ON \
-DAmesos_ENABLE_KLU=ON \
-DTrilinos_ENABLE_Amesos2=ON \
-DAmesos2_ENABLE_KLU2=ON \
-DAmesos2_ENABLE_Basker=ON \
-DTrilinos_ENABLE_Sacado=ON \
-DTrilinos_ENABLE_Stokhos=ON \
-DTrilinos_ENABLE_Kokkos=ON \
-DTrilinos_ENABLE_ALL_OPTIONAL_PACKAGES=OFF \
-DTrilinos_ENABLE_CXX11=ON \
-DTPL_ENABLE_AMD=ON \
-DAMD_LIBRARY_DIRS="/usr/lib" \
-DTPL_AMD_INCLUDE_DIRS="/usr/include/suitesparse" \
-DTPL_ENABLE_BLAS=ON \
-DTPL_ENABLE_LAPACK=ON \
..
```

Xyce

`cmake -D CMAKE_INSTALL_PREFIX=/usr/local -D Trilinos_ROOT=/usr/local -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_COMPILER=g++ -D Xyce_PLUGIN_SUPPORT=ON ..`

KrzysztofHerman, 2025-08-19:
@tvrusso thank you for the information. Probably you are right however, even though I have `-D Xyce_PLUGIN_SUPPORT=ON` in my `cmake` config I can not find any 'buildxyceplugin.sh' in my `/usr/local`. I have `admsXml` installed. Am I missing something ?

KrzysztofHerman, 2025-08-19:
@tvrusso 
I have no  `admsXml found TRUE`
but I got this one
`Plugin compatibility enabled`
in my cmake output. 
I got also some compilation issues/ Could you please share your cmake config for Trilinos to be 1:1 compatible ?

dwarning, 2025-08-21:
> 
> If you download the Xyce regression suite from https://github.com/Xyce/Xyce_Regression and change your Xyce CMake invocation to add `-DXyce_REGRESSION_DIR=/where/you/saved/Xyce_Regression`, then you can run all the tests from your build directory with `ctest -j8 -L "nightly" -LE "required:pymi"` (use as many processors after -j as you actually want, I have an 8-core machine and this speeds things up enormously). You will find the process running several "VERILOGPLUGIN" tests, each of which is verifying that some aspect of the plugin system is still working. I find they still are.

I added the define -DXyce_REGRESSION_DIR=/where/you/saved/Xyce_Regression in my cmake invocation. Then I started make (not cmake --build) which builded everything fine. But for

 `ctest -j4 -L "nightly" -LE "required:pymi"` in my Xyce/build dir

I got "No tests were found"
Should I use cmake build instead?

tvrusso, 2025-08-21:
Also, my own CMake invocation has `-DBUILD_TESTING=ON`, which I think may be necessary for the process to add the regression suite properly.

tvrusso, 2025-10-03:
This issue has been open for a long time without comment.  Did you ever get yourself back to having Xyce function with your Verilog-A plugins?  Or are you still stuck?

KrzysztofHerman, 2025-10-04:
@tvrusso thank you for taking care. I have delegated this work to our new team member. Let me ask her for the progress. In fact Xyce installation is the most challenging tool to install in our flow.  Right now please keep the issue open.